trinketapp/3D-testing #3: Possible gimbal lock issue

davidhoness, 2017-02-08:
Steps to reproduce
- Enable 3D mode
- Reset to 0 - 0 - 0
- Move yaw to 90
- Pitch has now become roll and roll is still roll
- Move roll to 90
- Yaw has now become pitch

The correct behaviour would be that the axes of rotation never change, or lock with another axis.

davidhoness, 2017-02-08:
Related, I remember the pitch axis has a limited output range from the RTIMU library. I previously believed this was a bug and raised the issue [here](https://github.com/RPi-Distro/python-sense-hat/issues/29). The author of the RTIMU library posted a link on there with some information about why the pitch axis had a limited range.

davidhoness, 2017-02-15:
By the way, here is an example where it's using the maths to avoid gimbal lock, if it's any use.
https://github.com/astro-pi/apollo-soyuz/blob/master/soyuz.py#L37

waywaaard, 2017-02-15:
Thanks for the information sharing. We do have a version without gimbal lock but are facing now some rendering bugs -.-

eah13, 2017-03-05:
Documenting some offline discussion: some amount of gimbal lock is endemic to roll/pitch/yaw inputs.  The drag-based inputs, which use quaternions, don't display this behavior, and always produce accurate Euler angle reads on the sliders, so that's a win.  

We've made an attempt to mitigate its effects the best we can and a decision to accept the remaining gimbal lock as practically unavoidable.  Avoiding gimbal lock altogether by converting the slider inputs to quaternions instead of Euler angles would create worse cognitive problems than the technical problems it solves.

Closing.  If there are suggestions for better mitigation math, seems best to open a separate issue for that once the latest 3D model is deployed (should be very soon).

closed 2017-03-05 by eah13