trinketapp/3D-testing #4: Feedback Requested: keyboard to joystick mapping

eah13, 2017-02-14:
Howdy testers!

We'd like specific feedback on the key-to-joystick mappings.  Currently, the 'up' key on your keyboard maps to the joystick state that sense_hat fires an 'up' keyboard event.  This means if you're at yaw=180 and press up in your browser, the joystick will move downwards, towards you.  

There are a few options:

* Leave as-is and accept the complexity of the display but stick with the simplicity of the up key on your keyboard always firing an 'up' event.
* When a trinket has the Astro Pi enclosure, the default yaw=90, so it's right side up.  We could map the 'up' key to the visual up direction in this mode, which would fire a 'left' event in the sense_hat module.  This is an 80% or compromise solution, covering the two most common cases (sense hat at yaw=0 and astro pi at yaw=90) intuitively by making 'up' move the joystick visually up.  
* We could make the joystick effect fully relative, so that pressing 'up' on your keyboard always moves the joystick northwards, whatever that is. There's some added complexity here, especially when the model is on its side or upside down, but visually this is the most simple.

Regardless of what we pick, We'd rely on tutorials to explain the underlying concepts, but the goal here is to choose the least bad cognitive dissonance.

rik-cross, 2017-02-17:
Hi,

I think our preference is the first option above -- i.e. stick with the simplicity of the 'up' key always firing an 'up- event.

Changing the mappings depending on the Sense HAT orientation could get very complicated, especially when the key/event mapping changes _during_ the the execution of a program. It may also get confusing for certain situations, e.g. what is the expected behaviour when the yaw is 135?

I've been told that when coding the Astro Pi, children come to realise that the 90 degree rotation of the Sense HAT means that pressing what looks like up actually fires a 'left' event. This is actually quite a nice learning opportunity, and the orientation of the Sense HAT feels like an extension of this same problem.

Sticking with the above also means that the emulator has a clearer parallel to what happens with the physical Sense HAT.

Rik.

eah13, 2017-03-16:
Closing! We went with 'up' always firing an 'up' event.  This means students will need to learn/visualize how the Sense HAT is oriented in the Astro Pi, which is facilitated by the enclosure toggling.

closed 2017-03-16 by eah13