trinketapp/3D-testing #5: Stub out pygame.display.set_mode

davidhoness, 2017-02-15:
API reference [here](https://www.pygame.org/docs/ref/display.html#pygame.display.set_mode).

```python
screen_width=640
screen_height=480
surface = pygame.display.set_mode([screen_width, screen_height])
```

In pygame a display [surface](https://www.pygame.org/docs/ref/surface.html) is required for the code to capture input device events. Under X-Windows this creates a new SDL window, of the width and height specified, which then must have focus to receive events. The same code without X-Windows will create a full-screen display which will automatically yield events. For Astro Pi the kids do this but leave the display blank since the flight units are headless on board the ISS.

I would be happy with just stubbing out the display and surface classes with NotImplementedError exceptions on everything except for the following:

display
- set_mode 
  - A call to set_mode should be enforced if they call into any of the [event](https://www.pygame.org/docs/ref/event.html) methods without having created a surface. That should avoid issues with porting code back and forth between a Pi and the Emulator.

surface
- get_width
- get_height
- get_size
- get_rect

So they can at least read back what they've sent in.

eah13, 2017-02-16:
@davidhoness a decision to make on this: should we make our pygame event handling dependent upon the existence of a surface? This won't pop up a window like it does locally but that should ensure that Trinket code runs locally.

davidhoness, 2017-02-16:
Yes please, as mentioned above, make it throw a warning or Exception if they call into any of the event methods if a surface hasn't been created yet.

eah13, 2017-02-16:
Ah, got it.  Misunderstood you.  Thanks!

eah13, 2017-03-12:
@davidhoness how important is `get_rect`? There's a bit of a rabbit hole to implement the `Rect` class, and I'm not sure that the returned object will be able to be manipulated in a meaningful way without support for creating an actual SDL surface.

eah13, 2017-03-16:
This is live in production, with the exception of get_rect.  Closing for now.  We can reopen if the `Rect` class is deemed to be core.

closed 2017-03-16 by eah13