trinketapp/3D-testing #6: three.js considerations

davidhoness, 2017-02-16:
Hi folks, this [stackoverflow](http://stackoverflow.com/questions/11243689/prefered-3d-model-format-of-three-js) article might be of interest to you regarding model formats.

Please let me know asap if you need us to undertake 3D modelling work.

eah13, 2017-02-16:
Thanks for finding this! We'll definitely need some help with modelling. For now we're using something that looks like this for mock-up purposes:

[image]

It was created from teidiv.com and is pure HTML/CSS. We're trying to avoid WebGL if we can since SVGs get bitmapped and ourt logic won't work with them. Can hop on a quick call letter today if that's helpful. We may be able to model ourselves but at the moment it's looking like getting a designer to build something in tridiv would be ideal.

eah13, 2017-02-16:
I made the above in 5 minutes with a ruler and a Sense HAT, for reference.

eah13, 2017-03-16:
three.js + tridiv model deployed! closing this.

closed 2017-03-16 by eah13