trinketapp/3D-testing #7: Fire Fox 3D CSS rendering

davidhoness, 2017-03-10:
[image]

I'm aware the TriDiv doesn't support Fire Fox and this is probably why, it has depth drawing problems. Can you let me know if anything can be done about this? If not we will probably have to make the decision to not support FF. The rendering performance was also pretty slow compared to Chrome on the same machine.

eah13, 2017-03-10:
Researching whether there are any workarounds.  [This article](https://css-tricks.com/things-watch-working-css-3d/) (ctrl-F for Firefox) has a minimal example and claims that FF just plain doesn't get 3d shape ordering right.  I can't find much on the Web about this, which surprises me

eah13, 2017-03-12:
As per our discussion, we'll create a flat/simple fallback for Firefox and probably Edge with a message to upgrade to a browser that supports 3d for the full experience.

waywaaard, 2017-03-15:
Those seems to be relevant to our issue with FireFox and 3d transforms:

https://bugzilla.mozilla.org/show_bug.cgi?id=1286716
https://bugzilla.mozilla.org/show_bug.cgi?id=725299
https://bugzilla.mozilla.org/show_bug.cgi?id=1097464
https://bugzilla.mozilla.org/show_bug.cgi?id=689498

All of those bugs can are related to `preserve-3d` and plane ordering.

eah13, 2017-03-16:
Changed this issue title to reflect the planned strategy for addressing this: using a 'flat' 3d model fallback for FF + IE that side-steps the non-support for `preserve-3d`.  Since the bug is only visible when there are multiple 3D volumes, we need to limit it to only one.

If there is support for a 3d volume with depth, and we can get design assets for all 6 sides, we may be able to provide a single cuboid for this fallback mode instead of a flat 2d image in 3d space.

eah13, 2017-03-31:
This is implemented!

closed 2017-03-31 by eah13