trinketapp/3D-testing #8: Gimbal lock avoidance math

davidhoness, 2017-03-10:
I really like the red, green and blue axes lines on the 3D model now. Now that these have been added it makes the gimbal lock problem all the more visible with the sliders though.

What should happen is the model always rotates around one of the lines, what should never happen is that all three lines move together when using one of the sliders. I feel that it's important that we do everything possible to achieve this.

Previously I have written a 3D demo which shows an object on the screen and uses input from the Sense HAT IMU sensor. I'll describe the maths below, but [here is a link](https://github.com/astro-pi/apollo-soyuz/blob/master/soyuz.py) to the code.

It may also be worth noting that the pitch axes on the Sense HAT is limited to 180 degrees and I wonder if doing the same thing here would help.

- Get pitch, roll and yaw in radians
  - sin_y = math.sin(yaw)
  - cos_y = math.cos(yaw)
  - sin_p = math.sin(pitch)
  - cos_p = math.cos(pitch)
  - sin_r = math.sin(roll)
  - cos_r = math.cos(roll)
- Calculate absolute roll
  - abs_roll = math.degrees(math.asin(sin_p * cos_y + cos_p * sin_r * sin_y))
- Calculate absolute pitch
  - abs_pitch = math.degrees(math.asin(sin_p * sin_y - cos_p * sin_r * cos_y))
- Rotate the model to absolute pitch and roll, and unmodified yaw
  - model.rotateToZ(abs_roll)
  - model.rotateToX(abs_pitch)
  - model.rotateToY(math.degrees(yaw))

eah13, 2017-03-10:
As per our discussion, we'll both try this out and experiment with axis isolation for the drag interface. Thanks for keeping on with the research!

eah13, 2017-03-31:
We decided to abandon slider-based input as a strategy for mitigating this. Closing for now.

closed 2017-03-31 by eah13