trinketapp/3D-testing #9: Cosmetic changes to 3D models

davidhoness, 2017-03-10:
**Raspberry Pi and Sense HAT**
- 4 corners of the Sense HAT, add 4 grey cylinders in to represent the stand-offs that secure the Sense HAT to the Raspberry Pi
- Add textures to all ports, USB x 2, LAN, AV, HDMI and power
- Use the SVG previously provided for the bottom of the Raspberry Pi

**Astro Pi**
- Use slightly different shades of grey for the top and sides, currently it's hard to see where the edges are and this makes the shape look less well defined
- Side T bar bracket attachment, use a different shade of gun mental grey than the main case
- Top side 4 button quad, move this upwards slightly. It's a bit too close to the bottom button pair
- Reduce diameter of joystick cap slightly, it should be about the same size as the hole for the humidity sensor just to the right and smaller than the push buttons
- Add heat sink pillars
- Bottom, the texture needs to be rotated through 180 degrees so that the SD card slot on the side matches up with the graphic on the bottom (not sure if you still need this after adding the pillars)
- Lighten the dark grey colours on the edge of the corner bolt enclosures

**Both**
- Use SVGs for textures as opposed to PNGs

eah13, 2017-03-12:
Thanks for these!

> Add textures to all ports, USB x 2, LAN, AV, HDMI and power

This is an area where we'll need a design asset.  We can do the surgery if you have an SVG of the relevant side views.  I don't recall seeing one yet but I think you mentioned they exist.

eah13, 2017-03-16:
These fixes (and a few others, like textures for the top of the Pi and bottom of the sense hat and the Astro bi top/bottom enclosure line) should be deployed in production! Please comment here if anything seems amiss.

closed 2017-03-16 by eah13