trinketapp/3D-testing #11: Yaw (and Compass) when flipping Sense HAT

tracygardner, 2017-03-31:
get_compass is tied to yaw. (Not sure if this is still an issue in the new UI as there's another issue with yaw.)

If we flip the Sense HAT 180 in the pitch and roll orientations then the Sense HAT is 'upside down' so I'd expect it to be rotated 180 via yaw (and therefore facing South) but it reports North. 

[image]

eah13, 2017-03-31:
Thanks for reporting! This is indeed a bug in our IMU implementation.  Looking into it now.

davidhoness, 2017-04-12:
@tracygardner can you retest?

eah13, 2017-04-14:
This is fixed in production

[image]

closed 2017-04-14 by eah13