trinketapp/3D-testing #12: Rotation doesn't seem right

davidhoness, 2017-04-10:
It still seems like the pitch, roll and yaw angles are tied to world rotation as opposed to the model rotation.
Steps to reproduce an example:

Reset the model to the home position using the double-arrow button.
[image]
Not sure why this isn't 0, 0, 0? On Astro Pi mode it seems to be 0, 0, 180 too. I thought that was supposed to be 0, 0, 90.

Anyway, then move the USB ports to the top.
[image]

Now pitch the SD card slot up and the USB ports down. We're doing a pitch movement but the roll is what is changing.
[image]

eah13, 2017-04-12:
The math behind the values calculation is [here](https://github.com/trinketapp/traqball.js/blob/master/src/traqball.js#L287). I believe the correct trig at that point should fix this.

eah13, 2017-04-14:
The fix is deployed!

[image]

[Rotated inverse matrix](https://github.com/trinketapp/traqball.js/blob/master/src/traqball.js#L282) was the answer.  *Obviously*.

Thanks a ton for the testing on this.  If there are any further problems please comment here and we'll investigate.

davidhoness, 2017-04-18:
I've had a look at the fix and that does seem to have done the trick. However, I noticed something else which is similar but to do with the feature you added to make it easier to turn the device around the yaw axis.

Steps to reproduce

- Rotate the device to be side on, looking directly at the HDMI port
  [image]
- Use yaw-drag feature
  [image]
- You're doing a pitch movement, but yaw is what is changing

I wonder if we should make the yaw-drag feature only available when at a particular orientation?

eah13, 2017-04-18:
@davidhoness thanks for this.  I'm able to replicate in the emulator following your instructions.

Have you compared with the physical device?  I can reproduce this there as well, so this may just be revealing the limitations of the Euler representation.

davidhoness, 2017-04-19:
Ah fascinating, so maybe it is right. I also see this with the physical hardware.

eah13, 2017-04-20:
Great!  Closing since this replicates physical device behavior

closed 2017-04-20 by eah13