trinketapp/3D-testing #13: Default rotation for old trinkets is now wrong

davidhoness, 2017-04-12:
It looks like the compass fix to ensure that geomagnetic North is at the top of the screen has caused a rotation problem for old trinkets.

Old trinkets now show with the Sense HAT inverted.
Example here: https://trinket.io/embed/python/308a373b5c
Which is used in this educational resource: https://www.raspberrypi.org/learning/getting-started-with-the-sense-hat/worksheet/

Currently a manual fix for this is to load the trinket, reset it and re-share it to get a new ID. Since we have the Sense HAT trinket embedded into a lot of our resources and in many places it's preferable if this can be fixed in software meaning no manual intervention required.

The fix for this should also not affect trinkets which have been reset and re-shared already to fix this problem, some of the Code Club resources have done the manual fix already.

eah13, 2017-04-12:
Fix on the way for this...

eah13, 2017-04-12:
Fix is deployed! Feel free to re-open if this isn't fixed after a refresh.

No code changes needed to old-style trinkets.  Also, the re-shared trinkets should all load using their new position settings, so no further changes needed.

closed 2017-04-12 by eah13