trinketapp/3D-testing #15: Moving central drag-bar moves the 3D model

davidhoness, 2017-04-12:
Possibly related to #14 

When the central drag-bar is moved, the position of the 3D model is either reset or changes.

Steps to reproduce:

- Move the model to an oblique angle
  [image]
- Move the central drag-bar slightly
- The model is either reset or changes position
  [image]
- The pitch, roll and yaw labels do not update

eah13, 2017-04-12:
Thanks.

Note: this seems related to #14 since they both change font size.

eah13, 2017-04-12:
Fix is deployed! Feel free to re-open if this isn't fixed after a refresh.

closed 2017-04-12 by eah13