trinketapp/3D-testing #16: Gyroscope raw emulation not working

davidhoness, 2017-04-12:
Originally reported [here](https://www.raspberrypi.org/blog/sense-hat-emulator-upgrade/#comment-1281104) on the RPF blog.

Affects the [this API call](http://pythonhosted.org/sense-hat/api/#get_gyroscope_raw) and it's alternatives. The numbers returned show the angular velocity of the three axes measured in [radians per second](https://en.wikipedia.org/wiki/Radian_per_second).

Example trinket: https://trinket.io/embed/python/9258a212a0

When you run the trinket you can see that the measurements returned never appear to peak above zero when the 3D model is moved.

Here are some examples measurements taken from a real Astro Pi, from top down I tilted it so I was looking directly down on the SD card slot, and the back to top down again a few moments later.

```
-0.033738	-0.080783	-0.009408
-0.053892	-0.010838	 0.022355
-0.089620	 0.008403	 0.076410
-0.083199	-0.042906	 0.058996
-0.053873	-0.087491	 0.006155
 0.002327	-0.070076	-0.017972
 0.007518	-0.001659	-0.007892
 0.000799	 0.016666	-0.007281
 0.007823	-0.005325	-0.003005
-0.000118	-0.038614	 0.009517
-0.026383	-0.029754	 0.006157
 0.056079	-0.000738	-0.000868
 0.000490	 0.016976	-0.003311
 0.011179	 0.038048	-0.002089
 0.004765	 0.022774	 0.006157
-0.008063	 0.093627	 0.012875
 0.012401	 0.213356	 0.009516
 0.040501	 0.273526	 0.010737
 0.052107	 0.217021	 0.021122
 0.008735	 0.124770	 0.021731
-0.008674	 0.110406	 0.020813
-0.006230	 0.232579	 0.012872
-0.008674	 0.314435	 0.009207
 0.000795	 0.352614	 0.010734
 0.031949	 0.240215	 0.021119
 0.010569	 0.079862	 0.043110
-0.043489	-0.063075	 0.048298
-0.083799	-0.173321	 0.010729
-0.023935	-0.241127	-0.018593
 0.101904	-0.710272	 0.008591
 0.232934	-2.656794	 0.112438
 0.135501	-4.887674	 0.420314
-0.363270	-8.510410	 0.296919
-0.946036	-10.025967	 0.253548
-1.595996	-10.039711	 0.522634
-2.198615	-10.039711	 0.872354
-1.872413	-10.039711	 0.651527
-1.274376	-8.298134	-0.487126
-1.008039	-2.105488	-0.975513
-0.849519	-0.034655	-1.203977
-0.663816	-0.488528	-0.761405
-0.466506	-1.440867	-0.041195
-0.192533	-1.057854	-0.066240
-0.225826	 0.286660	-0.281876
-0.103042	 0.505350	-0.356096
-0.074025	 0.361797	-0.228425
 0.059449	 0.260087	-0.109917
 0.045093	 0.346525	-0.054328
 0.089076	 0.153491	 0.026917
 0.061887	 0.089038	 0.032717
 0.052724	 0.315669	-0.032951
 0.016988	 0.348045	 0.001258
-0.014471	 0.318418	 0.015308
-0.019969	 0.165701	 0.011642
-0.023632	 0.042608	-0.014929
-0.051728	-0.009925	-0.023479
-0.040729	 0.075284	-0.001793
-0.014155	 0.065810	 0.030580
-0.007435	-0.091174	 0.021415
-0.029423	-0.114683	 0.013168
-0.032780	 0.074679	-0.014320
-0.038580	 0.067648	-0.032643
-0.027888	-0.142477	 0.033022
-0.004064	-0.123225	 0.047373
-0.016587	 0.064922	 0.004918
-0.039185	 0.139130	-0.014323
-0.026660	 0.088421	 0.086768
-0.019939	-0.137588	 0.065077
-0.031848	-0.126582	 0.084006
-0.021767	-0.081065	 0.075753
-0.038868	-0.027001	 0.044596
-0.065741	 0.122650	 0.016494
-0.037943	-0.003188	 0.056196
-0.006484	-0.203552	 0.136219
 0.001763	-0.111607	 0.090702
-0.006789	 0.129685	 0.065657
-0.007095	 0.189855	 0.066878
 0.025584	 0.034082	 0.027781
 0.006342	-0.013565	 0.011592
-0.007708	 0.083861	 0.023196
 0.002982	 0.028270	 0.086414
-0.011067	-0.121687	 0.043344
-0.026947	-0.042577	 0.029597
-0.028166	 0.078674	-0.027211
-0.033661	 0.094854	-0.017741
-0.013807	 0.027351	 0.024406
-0.011973	-0.129936	 0.047615
-0.022662	-0.074647	 0.016460
-0.014414	 0.066763	-0.004615
-0.003418	 0.002622	 0.026537
-0.003418	-0.115876	 0.068681
-0.004028	 0.167260	 0.063183
-0.064809	 0.748498	 0.027142
-0.225467	 3.243272	-0.069374
-0.211112	 8.060555	-0.493926
 0.021323	 9.936828	-0.744686
 0.243067	 9.975618	-0.734301
 0.742754	 9.975618	-0.757820
 0.897609	 9.975618	-0.178108
 1.014589	 8.966164	 0.157257
 0.713738	 0.479717	 1.260174
 0.052171	-2.846138	 1.363715
-0.272504	-1.893799	 0.864638
-0.229132	 0.035313	 0.572339
 0.052477	 0.965966	 0.174361
 0.137387	 0.404887	 0.090978
 0.021628	-0.112516	 0.161533
-0.030906	-0.237744	 0.207347
-0.068475	 0.024012	 0.128851
-0.034569	 0.068599	 0.071119
-0.053501	-0.043185	-0.001269
-0.103287	-0.103050	 0.066843
-0.082205	-0.145493	 0.070807
-0.064179	-0.083789	 0.079353
-0.058982	-0.012011	 0.043919
-0.055923	-0.004375	 0.011237
-0.045534	 0.055485	-0.022054
-0.050417	 0.090603	-0.043430
-0.062323	-0.001943	-0.031516
-0.057737	-0.178468	 0.010633
-0.050708	-0.153105	 0.020100
-0.070250	-0.193711	-0.008305
-0.088874	-0.183616	-0.012580
-0.082147	 0.039346	-0.048922
-0.071146	 0.147152	-0.031510
-0.060450	 0.100717	-0.019291
-0.057391	-0.026035	 0.014611
```

davidhoness, 2017-04-27:
Has this been deployed to live yet?

eah13, 2017-04-28:
Yes - sorry I forgot to put a comment here.  Production should be exactly as we tested it on dev in a recent call.  Note, of course, the limitations of not having actual HW access, but the emulator's readings will now jump to >5 in the case of fast movements.

closed 2017-04-28 by eah13