trinketapp/3D-testing #17: Google Chrome on Linux rotation not working

MarcScott, 2017-04-12:
The SenseHat can't be rotated using Google Chrome on Linux.
- OS Ubuntu
- 64bit Google Chrome installed using .deb

eah13, 2017-04-12:
Thanks for the report.  Could you share the version of Chrome you're using? We've had a report of rotation not working on Chromium 53 but that was not reproducible.

Chromium 57 on a debian derivative checks out, as does Chromium 56 on Raspbian.  If you have any plugins or other details to share, that would be helpful, as it seems like it may be system-specific in some way.

MarcScott, 2017-04-12:
`google-chrome --version` reports:
`Google Chrome 57.0.2987.133`

The rotation fails using guest mode as well, with all extensions
therefore disabled.

Ubuntu version is 16.10 (yakkety)

Elliott Hauser writes:

> Thanks for the report. Could you share the version of Chrome you're using? We've had a report of rotation not working on Chromium 53 but that was not reproducible.
>
> Chromium 57 on a debian [...]

eah13, 2017-04-12:
Unable to repro with Chrome 57.0.2987.133 on Gallium OS (derived from Xubuntu).

Thanks for sticking with me on this.  I'd love to try and identify the variables at play

**Specific Trinket**
Which trinket or page is this happening on for you?  I'll try to repro there.

**Cross Domain** (Don't know why this might affect it but good to check)
Does this happen for you here? https://trinket.io/sense-hat
What about on the [RPi blog post](https://www.raspberrypi.org/blog/sense-hat-emulator-upgrade/)?

**Mouse Position** (Sanity check)
The upper and lower corners of the output aren't active drag areas.  Are you able to rotate if you click near the center of the 3D model?

**Console Info**
Do you see any console errors? More tools.. > Developer tools > Console

MarcScott, 2017-04-13:
I can't reproduce this error on another laptop, also running Ubuntu and Chrome 57, so it looks like this is something unique to my current setup. This could take a disproportionate amount of time to investigate, considering the probable minuscule number of users it might affect. You might be better off just closing the issue.

However, as requested:

*Specific Trinket*
The error occurs on all trinkets

*Cross Domain*
The error occurs regardless of the domain the trinket is embedded in, and also on the trinket.io site.

*Mouse Position*
Clicking near the centre has no effect

*Console Info*
The only error is pasted below
```
jquery.min.js:4 Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help, check https://xhr.spec.whatwg.org/.
send	@	jquery.min.js:4
```

I've experimented with using Tridev - and I can rotate examples on their site [like this one](http://tridiv.com/app/index.html?dmlldz1wcmV2aWV3JmRvYz17InNldHRpbmdzIjp7ImxpZ2h0Ijoic3RhdGljIiwiYm9yZGVyIjoiMSIsImJnIjoiIzA4MTgyZiIsInNoYWRlIjoiLjQiLCJ0aW50IjoiMC4xOCIsInpvb20iOiIxMDAiLCJtYXRjaCI6IjAuNSIsInNuYXAiOiJvbiIsIm5hbWUiOiJYLXdpbmcifSwic2hhcGVzIjpbeyJ0eXBlIjoiY3Vib2lkIiwieSI6IjAuNjI1IiwieiI6MS43NSwidyI6My41LCJoIjoiMy4yNSIsImQiOjcuNSwiY0FsbCI6IiNiZmM5YzUiLCJuYW1lIjoiY3ViLTEifSx7InR5cGUiOiJwcmlzbSIsInkiOi0xLjUsInoiOjMuMjUsInciOjMuNSwiaCI6Ii45IiwiZCI6IjQuNCIsIm8iOiIuNSIsImNBbGwiOiIjMzI4NGNlIiwiY0JrIjoiIzAyMTAxZiIsImNCbSI6IiMxMjNhNWYiLCJuYW1lIjoicHJpLTEifSx7InR5cGUiOiJjdWJvaWQiLCJ5IjotMS41LCJ6IjotMC41LCJ3IjozLjUsImgiOjEsImNBbGwiOiIjYmZjOWM1IiwibmFtZSI6ImN1Yi0yIn0seyJ0eXBlIjoicHJpc20iLCJ5IjoiLTAuNjI1IiwieiI6MTAuNSwidyI6MiwiaCI6Ii43NSIsImQiOjEwLCJjQWxsIjoiI2JmYzljNSIsIm5hbWUiOiJwcmktMiJ9LHsidHlwZSI6InByaXNtIiwieCI6MS4yNSwieSI6MC41LCJ6IjoxMC41LCJ3IjoxLjUsImgiOjEsImQiOjEwLCJyeiI6IjkwIiwiY0Z0IjoiIzhiMjYyNiIsImNMdCI6IiNiZmM5YzUiLCJjUnQiOiIjYmZjOWM1IiwibmFtZSI6InByaS0zIn0seyJ0eXBlIjoicHJpc20iLCJ4IjotMS4yNSwieSI6MC41LCJ6IjoxMC41LCJ3IjoxLjUsImgiOjEsImQiOjEwLCJyeiI6MjcwLCJjQWxsIjoiI2JmYzljNSIsImNGdCI6IiM4YjI2MjYiLCJuYW1lIjoicHJpLTQifSx7InR5cGUiOiJwcmlzbSIsInkiOi0wLjI1LCJ6IjoxNywidyI6IjIiLCJoIjoiMSIsImNBbGwiOiIjYjFiYmI3IiwibmFtZSI6InByaS01In0seyJ0eXBlIjoiY3Vib2lkIiwieSI6MC43NSwieiI6MTcsInciOjIsImgiOiIxIiwiY0FsbCI6IiNiMWJiYjciLCJuYW1lIjoiY3ViLTMifSx7InR5cGUiOiJjdWJvaWQiLCJ4IjotNy43NSwieSI6LTEuNzUsInoiOi01LjUsInciOiIxMSIsImgiOiIuNSIsImQiOiI0IiwicnoiOiIxNiIsImNBbGwiOiIjYmZjOWM1IiwibmFtZSI6ImN1Yi00In0seyJ0eXBlIjoicHJpc20iLCJ4IjoyLjI1LCJ5IjotMC43NSwieiI6LTYuMjUsInciOjguNSwiaCI6IjIuNSIsImQiOjEsInJ5IjoiOTAiLCJjQWxsIjoiI2JmYzljNSIsImNMdCI6IiM2NzZmNzIiLCJuYW1lIjoicHJpLTYifSx7InR5cGUiOiJwcmlzbSIsIngiOi0yLjI1LCJ5IjotMC43NSwieiI6LTYuMjUsInciOjguNSwiaCI6IjIuNSIsImQiOjEsInJ5IjoiMjcwIiwicnoiOiIwIiwiY0FsbCI6IiNiZmM5YzUiLCJjUnQiOiIjNjc2ZjcyIiwibmFtZSI6InByaS03In0seyJ0eXBlIjoicHJpc20iLCJ4IjotMi4yNSwieSI6MS43NSwieiI6LTYuMjUsInciOjguNSwiaCI6IjIuNSIsImQiOjEsInJ4IjoiMTgwIiwicnkiOjI3MCwiY0FsbCI6IiNiZmM5YzUiLCJjTHQiOiIjNjc2ZjcyIiwibmFtZSI6InByaS04In0seyJ0eXBlIjoicHJpc20iLCJ4IjoyLjI1LCJ5IjoxLjc1LCJ6IjotNi4yNSwidyI6OC41LCJoIjoiMi41IiwiZCI6MSwicngiOiIxODAiLCJyeSI6IjkwIiwiY0FsbCI6IiNiZmM5YzUiLCJjUnQiOiIjNjc2ZjcyIiwibmFtZSI6InByaS05In0seyJ0eXBlIjoiY3lsaW5kZXIiLCJ4IjotMy41LCJ5IjotMiwieiI6LTUsImRpYW1ldGVyIjoiMi41IiwicngiOiItOTAiLCJzaWRlcyI6IjgiLCJjQWxsIjoiI2E2YWZhYyIsImNUcCI6IiMyZDMxMzMiLCJuYW1lIjoiY3lsLTEifSx7InR5cGUiOiJjeWxpbmRlciIsIngiOi0zLjUsInkiOi0yLCJ6IjotMTAsImRpYW1ldGVyIjoiMS41IiwicngiOiItOTAiLCJzaWRlcyI6IjYiLCJjQWxsIjoiIzdhNzE2YiIsIm5hbWUiOiJjeWwtMiJ9LHsidHlwZSI6InByaXNtIiwieCI6LTguNzUsInkiOi0yLCJ6IjotOC41LCJ3IjowLjUsImgiOiI5IiwiZCI6MiwicnkiOjE4MCwicnoiOiI3NCIsImNBbGwiOiIjYmZjOWM1IiwibmFtZSI6InByaS0xMCJ9LHsidHlwZSI6ImN5bGluZGVyIiwieCI6LTEyLjI1LCJ5IjotNCwieiI6LTUuNSwiZGlhbWV0ZXIiOiIxLjUiLCJoIjoiNiIsInJ4IjoiLTkwIiwicnkiOiIxIiwic2lkZXMiOiI2IiwiY0FsbCI6IiM4Yjk1OWEiLCJjVHAiOiIjNGY0ZjRmIiwibmFtZSI6ImN5bC0zIn0seyJ0eXBlIjoiY3Vib2lkIiwieCI6LTEyLjI1LCJ5IjotNCwieiI6MiwidyI6IjAuNCIsImgiOiIwLjQiLCJkIjoiOSIsImNBbGwiOiIjYjFiYmI3IiwibmFtZSI6ImN1Yi01In0seyJ0eXBlIjoiY3Vib2lkIiwieCI6Ny43NSwieSI6LTEuNzUsInoiOi01LjUsInciOiIxMSIsImgiOiIuNSIsImQiOiI0IiwicngiOiItMTgwIiwicnoiOiIxNiIsImNBbGwiOiIjYmZjOWM1IiwibmFtZSI6ImN1Yi02In0seyJ0eXBlIjoiY3lsaW5kZXIiLCJ4IjozLjUsInkiOi0yLCJ6IjotNSwiZGlhbWV0ZXIiOiIyLjUiLCJyeCI6Ii05MCIsInNpZGVzIjoiOCIsImNBbGwiOiIjYTZhZmFjIiwiY1RwIjoiIzJkMzEzMyIsIm5hbWUiOiJjeWwtNCJ9LHsidHlwZSI6ImN5bGluZGVyIiwieCI6My41LCJ5IjotMiwieiI6LTEwLCJkaWFtZXRlciI6IjEuNSIsInJ4IjoiLTkwIiwic2lkZXMiOiI2IiwiY0FsbCI6IiM3YTcxNmIiLCJuYW1lIjoiY3lsLTUifSx7InR5cGUiOiJjeWxpbmRlciIsIngiOjEyLjI1LCJ5IjotNCwieiI6LTUuNSwiZGlhbWV0ZXIiOiIxLjUiLCJoIjoiNiIsInJ4IjotOTAsInJ5IjoiLTEiLCJzaWRlcyI6IjYiLCJjQWxsIjoiIzhiOTU5YSIsImNUcCI6IiMzNTM1MzUiLCJuYW1lIjoiY3lsLTYifSx7InR5cGUiOiJjdWJvaWQiLCJ4IjoxMi4yNSwieSI6LTQsInoiOjIsInciOiIwLjQiLCJoIjoiMC40IiwiZCI6IjkiLCJjQWxsIjoiI2IxYmJiNyIsIm5hbWUiOiJjdWItNyJ9LHsidHlwZSI6InByaXNtIiwieCI6OC43NSwieSI6LTIsInoiOi04LjUsInciOjAuNSwiaCI6IjkiLCJkIjoyLCJyeCI6Ii0xODAiLCJyeiI6IjEwNiIsImNBbGwiOiIjYmZjOWM1IiwibmFtZSI6InByaS0xMSJ9LHsidHlwZSI6ImN1Ym9pZCIsIngiOi03Ljc1LCJ5IjoyLjc1LCJ6IjotNS41LCJ3IjoiMTEiLCJoIjoiLjUiLCJkIjoiNCIsInJ6IjoiLTE2IiwiY0FsbCI6IiNiZmM5YzUiLCJuYW1lIjoiY3ViLTgifSx7InR5cGUiOiJjdWJvaWQiLCJ4Ijo3Ljc1LCJ5IjoyLjc1LCJ6IjotNS41LCJ3IjoiMTEiLCJoIjoiLjUiLCJkIjoiNCIsInJ4IjoiLTE4MCIsInJ6IjoiLTE2IiwiY0FsbCI6IiNiZmM5YzUiLCJuYW1lIjoiY3ViLTkifSx7InR5cGUiOiJjeWxpbmRlciIsIngiOi0zLjUsInkiOjMsInoiOi01LCJkaWFtZXRlciI6IjIuNSIsInJ4IjoiLTkwIiwic2lkZXMiOiI4IiwiY0FsbCI6IiNhNmFmYWMiLCJjVHAiOiIjMmQzMTMzIiwibmFtZSI6ImN5bC03In0seyJ0e [... cut ...] jozLjUsInkiOjMsInoiOiItMi40MSIsInciOiIyLjUiLCJoIjoiMC4yNSIsImQiOiIwLjEiLCJyeiI6IjE2IiwiY0FsbCI6IiM5Nzk3OWIiLCJuYW1lIjoiY3ViLTE2In0seyJ0eXBlIjoiY3Vib2lkIiwieCI6LTguNSwieSI6Ii0yLjMiLCJ6IjotNS4yNSwidyI6IjIuNSIsImgiOiIuMSIsImQiOiIzLjUiLCJyeiI6IjE2IiwiY0FsbCI6IiNiZmM5YzUiLCJjVHAiOiIjOGIyNjI2IiwibmFtZSI6ImN1Yi0xNyJ9LHsidHlwZSI6ImN1Ym9pZCIsIngiOiI4LjUiLCJ5IjoiLTIuMyIsInoiOi01LjI1LCJ3IjoiMi41IiwiaCI6Ii4xIiwiZCI6IjMuNSIsInJ6IjoiLTE2IiwiY0FsbCI6IiNiZmM5YzUiLCJjVHAiOiIjOGIyNjI2IiwibmFtZSI6ImN1Yi0xOCJ9XX0=)

davidhoness, 2017-04-13:
Ah so this is a reproduction of the issue I reported on April 10 over email. Marc and I spoke this morning and it's the exact same problem as I've seen on Chromium version 53 on Ubuntu 14.04.

```
Chromium	53.0.2785.143 (Developer Build) Built on Ubuntu , running on Ubuntu 14.04 (64-bit)
Revision	12ae4ebf489873fc4aad9efc231b9151b330a938
OS	Linux 
Blink	537.36 (@12ae4ebf489873fc4aad9efc231b9151b330a938)
JavaScript	V8 5.3.332.47
Flash	(Disabled)
User Agent	Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/53.0.2785.143 Chrome/53.0.2785.143 Safari/537.36
Command Line	/usr/lib/chromium-browser/chromium-browser --enable-pinch --window-depth=24 --wm-user-time-ms=916747 --x11-visual-id=32 --window-depth=24 --x11-visual-id=32 --wm-user-time-ms=1116464 --flag-switches-begin --flag-switches-end
Executable Path	/usr/lib/chromium-browser/chromium-browser
Profile Path	/home/dave/.config/chromium/Default
Variations	16e0dd70-3f4a17df
ba3f87da-92cc81ec
f049a919-3f4a17df
dd4da2fc-3f4a17df
43d0dd1e-3f4a17df
64cbdfc2-3f4a17df
b7786474-d93a0620
868bda90-3f4a17df
4ea303a6-3f4a17df
9736de91-3f4a17df
dbffab5d-f51b51
30e679f-3f4a17df
ad6d27cc-3e870323
867c4c68-3f4a17df
d747916f-d747916f
fe05be5f-4ad60575
```

MarcScott, 2017-04-13:
My chrome information
```
Google Chrome	57.0.2987.133 (Official Build) (64-bit)
Revision	ec33cd0c06881d919ac0de419d829ad914e0be8f-refs/branch-heads/2987@{#887}
OS	Linux 
JavaScript	V8 5.7.492.71
Flash	25.0.0.148 /home/mjs/.config/google-chrome/PepperFlash/25.0.0.148/libpepflashplayer.so
User Agent	Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.133 Safari/537.36
Command Line	/usr/bin/google-chrome-stable --flag-switches-begin --flag-switches-end
Executable Path	/opt/google/chrome/google-chrome
Profile Path	/home/mjs/.config/google-chrome/Profile 1
Variations	3095aa95-3f4a17df
7c1bc906-f55a7974
2a33b90e-61faa026
ba3f87da-b4a760c3
cf558fa6-48a16532
f3499283-f94608c8
31362330-ca7d8d80
349d561b-3f4a17df
9e201a2b-26bbc519
6eb432aa-ca7d8d80
5274eb09-3f4a17df
684d1cdf-51126808
9773d3bd-ca7d8d80
b22b3d54-3d47f4f4
2e109477-ca7d8d80
9e5c75f1-53392e0f
6b121ae7-ca7d8d80
f79cb77b-3d47f4f4
b7786474-d93a0620
23a898eb-ca7d8d80
4ea303a6-ecbb250e
9736de91-ca7d8d80
64005e71-4ad60575
69bf80fa-91c810ef
867c4c68-3f4a17df
b2f0086-93053e47
7fc902e8-ca7d8d80
f11cb941-11910166
3ac60855-486e2a9c
f296190c-92a7f5c9
4442aae2-75cb33fc
ed1d377-e1cc0f14
75f0f0a0-6bdfffe7
e2b18481-3a9ae350
e7e71889-e1cc0f14```

davidhoness, 2017-04-13:
@eah13 would it be possible for you to make a build with debug console output on elliott-dev.trinket.io so we can at identify how far it's getting? For example, are the mouse events even firing and if they are how far do the calls get into the code which moves the model?

davidhoness, 2017-04-20:
Interestingly, I've also found the demo on [this page](http://w3.eleqtriq.com/2011/11/traqball-2-0/) doesn't work in my version of Chromium. Click *Open Example* and try to drag the stars. Is this the same for you @MarcScott?

MarcScott, 2017-04-21:
Also tested using [traqball](http://w3.eleqtriq.com/2011/11/traqball-2-0/) demo site and fails to work for me on Chrome, while working fine in Firefox.

davidhoness, 2017-04-21:
Folks, from the work Brian has done I've found the following config changes fix the problem.

- Open Chrome
- Type chrome://flags/ in the address bar and press Enter.
- Click on Enable touch events > Disabled.
- Click on Relaunch Now.

MarcScott, 2017-04-21:
Works for me. On Vanilla Chrome the option is `Touch Events API` that needs disabling.

davidhoness, 2017-04-21:
And as it turns out both @MarcScott and I have laptops with touch-screens, but with USB mice plugged in.

brianpmarks, 2017-04-21:
Thanks @davidhoness and @MarcScott for help testing!

We've got a fix for this pending in [trinketapp/traqball.js](https://github.com/trinketapp/traqball.js). We should be able to get the deployed relatively soon and we'll update this issue once it's live.

closed 2017-04-21 by eah13

davidhoness, 2017-04-24:
Is this live yet?

brianpmarks, 2017-04-24:
Apologies for the delay. This is now live.

davidhoness, 2017-04-24:
Nice one, thank you. Fix verified on my end.