trinketapp/3D-testing #19: Emulator home page iframe not be showing in Microsoft Edge

davidhoness, 2017-04-13:
It seems the iframe isn't loading on Edge / Windows 10

Steps to reproduce

- Open Microsoft Edge
- Navigate to https://trinket.io/sense-hat
- Iframe doesn't load
  [image]

I was using
Microsoft Edge 38.14393.0.0
Microsoft EdgeHTML 14.14393

davidhoness, 2017-04-13:
Works fine if you navigate directly to a trinket ID though.
[image]

waywaaard, 2017-04-13:
I can reproduce this error on windows 10 with Creators Update installed. The console output indicates some problems with the landing page as it tries to load `scss` files which are not available and should not be requested at all. Maybe there is something broken in der grunt tasks?

[image]

eah13, 2017-04-13:
@waywaaard thanks for this. I wonder if these 404s are related to sourcemap handling.  If so, they might be unrelated - do they happen on other pages as well?  Anything in the console besides those 404s?  I'll have access to edge later today to investigate.

brianpmarks, 2017-04-13:
Since it's only happening on the sense-hat page and not in regular iframes, I suspect it has to do with how we load the code on that particular page (via the query string.) Looking into it now.

eah13, 2017-04-14:
@davidhoness the fix for this is in production.  Please re-test at your convenience.

Brian discovered that apparently Edge has decided to limit URL length, which required a different way of loading trinkets on the demo page.

waywaaard, 2017-04-14:
I get those errors in Edge also by visiting this https://trinket.io/python/25f043a1a0 . However, the 2D version is loaded and displayed fine.

eah13, 2017-04-20:
Confirmed fixed

closed 2017-04-20 by eah13