trinketapp/3D-testing #20: Pygame event.key being returned as None in Microsoft Edge

davidhoness, 2017-04-13:
Example trinket: https://trinket.io/python/25f043a1a0

Steps to reproduce

- Open Microsoft Edge
- Navigate to https://trinket.io/python/25f043a1a0
- Press any key
- Note debug output, key property is of None type, unicode and mod properties are also not set.
  ```
  <Event(2-KeyDown {'unicode': '', 'key': None, 'scancode': 0, 'mod': 0})>
  <Event(3-KeyUp {'unicode': '', 'key': None, 'scancode': 0, 'mod': 0})>
  <Event(2-KeyDown {'unicode': '', 'key': None, 'scancode': 0, 'mod': 0})>
  <Event(3-KeyUp {'unicode': '', 'key': None, 'scancode': 0, 'mod': 0})>
  ```

Discovered while trying to play with [this trinket](https://trinket.io/embed/python/75fdd54c9c) on Edge.

I was using
Microsoft Edge 38.14393.0.0
Microsoft EdgeHTML 14.14393

eah13, 2017-04-13:
Thanks - looking into this.

albertjan, 2017-04-14:
I checked with Edge 25 and works 😕 , now updating my windows machine.

waywaaard, 2017-04-14:
I can reproduce this error in Edge 40 on Windows 10

eah13, 2017-05-05:
This should be fixed in production.  Thanks to @davidhoness for help testing

closed 2017-05-05 by eah13