trinketapp/3D-testing #22: FireFox User-Agent detection in Embedded trinkets

davidhoness, 2017-04-27:
Found an embedded trinket where FireFox gets the 3D model as opposed to the 2D one.
Example here: https://codeclubprojects.org/en-GB/sense-hat/tightrope/

eah13, 2017-04-28:
Reproduced on this and [other trinkets](https://codeclubprojects.org/en-GB/sense-hat/compass-maze/) embedded on codeclubprojects.org.  Will have to look into this.  I'm unsure exactly how this would happen, but it seems related to the domain the trinket is embedded in.  The [RPi blog post](https://www.raspberrypi.org/blog/sense-hat-emulator-upgrade/) is unaffected.

eah13, 2017-04-28:
Scratch that: it's related to the `outputOnly=true` query parameter; https://trinket.io/python/e13bde8f93?outputOnly=true

We'll look into a fix.

brianpmarks, 2017-05-01:
Thanks for this report. Firefox should now be getting the 2D version on these sites. Please reopen if still an issue.

closed 2017-05-01 by brianpmarks