trinketapp/3D-testing #25: Sense HAT emulator freezing

tracygardner, 2017-05-25:
We've had this issue reported on our live Sense HAT projects and need to work out what the correct fix is.

Projects freeze in some environments if we don't include a sleep in loops that use the Sense HAT api but don't do much else. E.g. https://trinket.io/library/trinkets/eaea4cb76c (remove the sleep to cause the issue.)

This wasn't the case previously (people who have previously successfully run the projects now can't.) I can't replicate it on Chrome Ubuntu though it has been replicated on Windows and Mac. 

It isn't necessary to add a sleep on a physical Sense HAT so the emulator should probably cope to be realistic. Thoughts?

(Apologies if this isn't the correct repo, but couldn't see a better one.)

brianpmarks, 2017-05-26:
Thanks for the report! I've been able to reproduce and am looking into it.

brianpmarks, 2017-05-26:
Thanks again for this report. We just made a change that should fix the freezing issue and the call to sleep can be safely removed. I'll close this issue but please re-open if you still see problems or hear more reports.

closed 2017-05-26 by brianpmarks